ethz-asl/ethzasl_msf #102: Set up help with lsd-slam

henn1001, 2015-02-22:
So i'm trying to get this framework to work with lsd-slam and i'm not quite sure if i set up everything correctly. 

I already rotated the output of my vslam so that it aligns with my imu as seen here (small one is the imu)
![screenshot from 2015-02-22 22 45 38]
(https://cloud.githubusercontent.com/assets/10023892/6320673/c9e7ebaa-bae4-11e4-9e39-9470a4c3704f.png)

I'm using the following launch file: 
[image]

and the following config file
[image]

So what i want/expect is a metric scaled position output ( set absolute_measurements: false)

I did not set up a covariance matrix in lsd slam. I think it just sends zeros. ( fixed_covariance: true) 
Is this correct? Or do i have  to send dummy values?

Not quite sure about the q_ic values but the pose output of the framework seems to be aligned. I should not have to rotate anything because my sensors are already aligned?!
Also do still have to set fixed p_ic and q_ic  to true? 

I'm asking all this because i see some mixed results in the accuracy of the position - ( they are really bad and often delayed).  Its better if i set absolute_mearurements= true but this is expected since it just hogs to the lsd unscaled position values.

Also i 'm getting this message here quite often:
[ WARN] [1424640218.513812822]: msf_core: imu message drop curr seq:263297 expected: 263296
[ WARN] [1424640218.824028297]: msf_core: imu message drop curr seq:263329 expected: 263328
[ WARN] [1424640219.164891690]: msf_core: imu message drop curr seq:263363 expected: 263362
I tried to change the delay value but it did not help.

simonlynen, 2015-02-23:
@henn1001 several points that you should take a close look at:
- you do need to provide the IMU to camera transformation. What the framework expects here is the position and orientation of the camera w.r.t. the IMU frame of reference, expressed in the IMU frame of reference. This is the first and most important part to fix.
- ```absolute measurements``` refers to the fact that your measurements are not a relative quantity but denote the position of the camera system wrt. some global base-frame. You will need to switch this parameters to ```true```
- don't fix the scale factor. Set ```pose_sensor/pose/fixed_scale``` to ```false``` (also you have this setting twice in your yaml, so you need to clean these up too)
- ```pose_sensor/pose_measurement_world_sensor``` needs to be double checked. If LSD slam provides the position/orientation of the camera wrt. a "world" frame of reference then this must be set to true. If LSD slam provides the position/orientation of a "world" frame expressed in the camera frame of reference, then this must be set to false.
- Yes, if the covariance of LSD is not set, then you should use the ```fixed-covariance``` setting in the estimator.

- The dropouts on the IMU you are seeing are probably due to your PC being too slow or your IMU node having in fact dropouts. If this is just about single messages, then it should work still fine.

henn1001, 2015-02-23:
Thanks for your reply and your help - really appreciate that 

- Yeah i understand this part. Since my Imu is sitting directly under my camera and the frames are already aligned, i wont have to rotate anything right? so i should be good with 1,0,0,0 / w,x,y,z ? i'm not that good with quaternions : )

- OK so if i set my position/vslam sensor as a absolute measurement - the thing that will be wrong is the scale. that's why i shouldn't use a fixed scale and where "pose_initial_scale:" comes into play.  But over time the scale should automatically converge to a metric scale right?

- "pose_measurement_world_sensor: false"  should be correct - same as ptam

- i fixed the imu dropouts -> had to set up a queue in my publisher node 

Also i did some testing:
[image]
This is the position output from lsd_slam. after the initialization i moved it back and forth ~40  cm (x axis) and then to the side ~60 cm (y axis)

[image]
MSF output with absolute measurements = true 

[image]
and here with a initial_scale = 0.5

art-mx, 2015-11-10:
Hi, @henn1001 Have you managed to get your implementation of msf setup to work?