ethz-asl/ethzasl_msf #104: Compilation Conflict With asctec_mav_framework Package

XinkeAE, 2015-03-19:
Hi,

Great work! Thanks for providing this great framework to the community!

I'm having trouble in installing the asctec_mav_framework. This is the process I ran through:

Got the "catkin_simple" package installed;

Got the "glog_catkin" package install;

Created a folder ethz_asctec_mav_framework(this can be arbitrary)/src;
Download the zip file and unzip inside the src folder;
Run "catkin_make" outside of the src;

The compilation system doesn't like this and just gave me the error:
****************************Error*******************************
"CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
Could not find a package configuration file provided by
"sensor_fusion_comm" with any of the following names:

sensor_fusion_commConfig.cmake
sensor_fusion_comm-config.cmake"
****************************EndError*******************************
From previous issue, I found this is because I didn't install the sensor_fusion package. I tried to run similar process with the "ethzasl_msf" package and found this error:
****************************Error*******************************
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
Could not find a package configuration file provided by "asctec_hl_comm"
with any of the following names:

asctec_hl_commConfig.cmake
asctec_hl_comm-config.cmake
****************************EndError*******************************
Obviously this is because the framework hasn't been installed.

Could you please give me some suggestions on this issue?

Thanks!
Xinke

omaris, 2015-03-19:
Hi Xinke,

The asctec_hl_comm package is part of asctec_mav_framework (
https://github.com/ethz-asl/asctec_mav_framework). If you downloaded the
whole repo in your workspace, this error should not have happened. What
kind of zip file did you download? Try cloning the following repos into
your catkin workspace (<catkin_ws>/src):

git clone git@github.com:ethz-asl/asctec_mav_framework.git
git clone git@github.com:ethz-asl/ethzasl_msf.git
git clone --recursive git@github.com:ethz-asl/thirdparty_submodules.git

This should automatically download all necessary packages required for msf.
To build catkin packages, I recommend using the catkin build tools (
http://catkin-tools.readthedocs.org/en/latest/verbs/catkin_build.html)
instead of catkin_make. One of the advantages is that you can compile
individual packages instead of your full workspace.

best

Sammy

markusachtelik, 2015-03-19:
closing this, let's fix everything in https://github.com/ethz-asl/asctec_mav_framework/issues/33

closed 2015-03-19 by markusachtelik