ethz-asl/ethzasl_msf #109: Inquiry about state output as ros/tf

prclibo, 2015-04-10:
Hi, about this line [L343](https://github.com/ethz-asl/ethzasl_msf/blob/master/msf_core/include/msf_core/msf_sensormanagerROS.h#L343)

    tf_broadcaster_.sendTransform(
        tf::StampedTransform(
            transform, ros::Time::now() /*ros::Time(latestState->time_)*/,
            "world", "state"));

I thought the stamp of the sent transform should be `state_const->time`. I would like to inquiry why `ros::Time::now()` is used. Thank you:)

simonlynen, 2015-04-10:
I think this was done for rviz since it drops messages when the filter runs from a bag file unless rviz runs in simulation time. But I would suggest we revert this since it is not correct.

prclibo, 2015-04-10:
OK, thanks for the explanation.