ethz-asl/ethzasl_msf #110: build problem: Eigen static assertion failed

Daniel-Eckert, 2015-04-23:
I have trouble building the msf_updates package. This is what I get as an error message:

    ...
    /home/daniel/catkin_ws/src/ethzasl_msf/msf_updates/src/position_msf/main.cpp:29:1:   required from here
    /usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
    /usr/include/eigen3/Eigen/src/Core/Assign.h:493:3: note: in expansion of macro EIGEN_STATIC_ASSERT
       EIGEN_STATIC_ASSERT(SameType,YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY)
       ^
    make[2]: *** [src/pose_msf/CMakeFiles/pose_sensor.dir/main.cpp.o] Error 1
    make[1]: *** [src/pose_msf/CMakeFiles/pose_sensor.dir/all] Error 2
    make[2]: *** [src/position_msf/CMakeFiles/position_sensor.dir/main.cpp.o] Error 1
    make[1]: *** [src/position_msf/CMakeFiles/position_sensor.dir/all] Error 2
    make[2]: *** [src/pose_pressure_msf/CMakeFiles/pose_pressure_sensor.dir/main.cpp.o] Error 1
    make[1]: *** [src/pose_pressure_msf/CMakeFiles/pose_pressure_sensor.dir/all] Error 2
    make: *** [all] Error 2
    [msf_updates] <== '/home/daniel/catkin_ws/build/msf_updates/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2'

Any idea what the issue could be here? All dependencies should be correct. 

Regards,
Daniel

simonlynen, 2015-04-23:
As the error message says, it seems you have added some code in line 29 of ```/home/daniel/catkin_ws/src/ethzasl_msf/msf_updates/src/position_msf/main.cpp``` in which you try to do Matrix operations with operants that have different types, i.e. multiply a double matrix with a float matrix.

Daniel-Eckert, 2015-04-23:
Hi Simon, Thanks for the quick reply!

I didn't change anything in the repository. It's cloned and still in it's original state. All other packages in the repository get built perfectly.

main.cpp looks like this:

    #include "position_sensormanager.h"
    int main(int argc, char** argv) {
      ros::init(argc, argv, "msf_position_sensor");
      msf_position_sensor::PositionSensorManager manager;
      ros::spin();
      return 0;
    }          // line 29

simonlynen, 2015-04-23:
Hi @Daniel-Eckert ,
what is the command you use to build the package?

Daniel-Eckert, 2015-04-24:
I use catkin build, and tried catkin_make, but both give exactly the same error.

simonlynen, 2015-04-24:
@Daniel-Eckert since I can't reproduce this on any machine here and also not on our continuous integration server (running Ubuntu 14.04 and OSX) I need more information:
- what is the platform you build on including the version of ROS.
- can you post the _full_ output of:
  ```
  cd ~/catkin_ws/build/msf_updates
  make
  ```

Daniel-Eckert, 2015-04-24:
@simonlynen I build on Ubuntu 14.04, and ROS indigo.

The full output of make this:

    daniel@Daniel:~/catkin_ws/build/msf_updates$ make
    -- Using CATKIN_DEVEL_PREFIX: /home/daniel/catkin_ws/devel
    -- Using CMAKE_PREFIX_PATH: /home/daniel/catkin_ws/devel;/opt/ros/indigo
    -- This workspace overlays: /home/daniel/catkin_ws/devel;/opt/ros/indigo
    -- Using PYTHON_EXECUTABLE: /usr/bin/python
    -- Using Debian Python package layout
    -- Using empy: /usr/bin/empy
    -- Using CATKIN_ENABLE_TESTING: ON
    -- Call enable_testing()
    -- Using CATKIN_TEST_RESULTS_DIR: /home/daniel/catkin_ws/build/msf_updates/test_results
    -- Found gtest: gtests will be built
    -- Using Python nosetests: /usr/bin/nosetests-2.7
    -- catkin 0.6.11
    -- Using these message generators: gencpp;genlisp;genpy
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /home/daniel/catkin_ws/build/msf_updates
    [ 41%] Built target msf_updates_gencfg
    [ 50%] Building CXX object CMakeFiles/pose_distorter.dir/src/msf_distort/PoseDistorter.cc.o
    Linking CXX shared library /home/daniel/catkin_ws/devel/lib/libpose_distorter.so
    [ 50%] Built target pose_distorter
    [ 58%] Building CXX object CMakeFiles/test_distort.dir/src/test/test_distort.cc.o
    Linking CXX executable /home/daniel/catkin_ws/devel/lib/msf_updates/test_distort
    [ 58%] Built target test_distort
    [ 66%] Building CXX object src/pose_msf/CMakeFiles/pose_sensor.dir/main.cpp.o
    In file included from /usr/include/eigen3/Eigen/Core:328:0,
                     from /usr/include/eigen3/Eigen/Dense:1,
                     from /usr/include/eigen3/Eigen/Eigen:1,
                     from /home/daniel/catkin_ws/src/ethzasl_msf/msf_core/include/msf_core/msf_core.h:39,
                     from /home/daniel/catkin_ws/src/ethzasl_msf/msf_updates/src/pose_msf/pose_sensormanager.h:22,
                     from /home/daniel/catkin_ws/src/ethzasl_msf/msf_updates/src/pose_msf/main.cpp:19:
    /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h: In instantiation of ‘struct Eigen::internal::traits<Eigen::CoeffBasedProduct<const Eigen::Matrix<long double, 3, 6>&, const Eigen::Matrix<double, 6, 6>&, 6> >’:
    /usr/include/eigen3/Eigen/src/Core/DenseBase.h:41:34:   required from ‘class Eigen::DenseBase<Eigen::CoeffBasedProduct<const Eigen::Matrix<long double, 3, 6>&, const Eigen::Matrix<double, 6, 6>&, 6> >’
    /usr/include/eigen3/Eigen/src/Core/MatrixBase.h:48:34:   required from ‘class Eigen::MatrixBase<Eigen::CoeffBasedProduct<const Eigen::Matrix<long double, 3, 6>&, const Eigen::Matrix<double, 6, 6>&, 6> >’
    /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:113:7:   required from ‘class Eigen::CoeffBasedProduct<const Eigen::Matrix<long double, 3, 6>&, const Eigen::Matrix<double, 6, 6>&, 6>’
    /home/daniel/catkin_ws/src/ethzasl_msf/msf_core/include/msf_core/implementation/msf_state_inl.h:419:86:   required from here
    /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:43:112: error: no type named ‘ReturnType’ in ‘struct Eigen::internal::scalar_product_traits<long double, double>’
    typedef typename scalar_product_traits<typename _LhsNested::Scalar, typename _RhsNested::Scalar>::ReturnType Scalar;
                                                                                                                ^
    /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:69:51: error: no type named ‘ReturnType’ in ‘struct Eigen::internal::scalar_product_traits<long double, double>’
                           || ( (ColsAtCompileTime % packet_traits<Scalar>::size) == 0
                                                   ^
    /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:76:51: error: no type named ‘ReturnType’ in ‘struct Eigen::internal::scalar_product_traits<long double, double>’
                           || ( (RowsAtCompileTime % packet_traits<Scalar>::size) == 0
                                                   ^
    /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:94:41: error: no type named ‘ReturnType’ in ‘struct Eigen::internal::scalar_product_traits<long double, double>’
                       + (InnerSize - 1) * NumTraits<Scalar>::AddCost,
                                         ^
    /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:93:63: error: no type named ‘ReturnType’ in ‘struct Eigen::internal::scalar_product_traits<long double, double>’
                     : InnerSize * (NumTraits<Scalar>::MulCost + LhsCoeffReadCost + RhsCoeffReadCost)
                                                               ^
    /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:106:41: error: no type named ‘ReturnType’ in ‘struct Eigen::internal::scalar_product_traits<long double, double>’
                           && (InnerSize % packet_traits<Scalar>::size == 0)
                                         ^
    In file included from /usr/include/eigen3/Eigen/Core:278:0,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /usr/include/eigen3/Eigen/Eigen:1,
                 from /home/daniel/catkin_ws/src/ethzasl_msf/msf_core/include/msf_core/msf_core.h:39,
                 from /home/daniel/catkin_ws/src/ethzasl_msf/msf_updates/src/pose_msf/pose_sensormanager.h:22,
                 from /home/daniel/catkin_ws/src/ethzasl_msf/msf_updates/src/pose_msf/main.cpp:19:
[... cut ...]
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
    /usr/include/eigen3/Eigen/src/Core/Assign.h:493:3: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
    EIGEN_STATIC_ASSERT(SameType,YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY)
     ^
    make[2]: *** [src/pose_msf/CMakeFiles/pose_sensor.dir/main.cpp.o] Error 1
    make[1]: *** [src/pose_msf/CMakeFiles/pose_sensor.dir/all] Error 2
    make: *** [all] Error 2

simonlynen, 2015-04-24:
@Daniel-Eckert can you try out https://github.com/ethz-asl/ethzasl_msf/pull/111

Daniel-Eckert, 2015-04-24:
Works perfectly. Thanks a lot!

closed 2015-04-24 by Daniel-Eckert

amjack0, 2020-08-24:
can anyone post the found solution ?

simonlynen, 2020-08-26:
This should have been fixed by https://github.com/ethz-asl/ethzasl_msf/pull/111 a long time ago. Can you open a new issue with the compilation error (and full output) please?