ethz-asl/ethzasl_msf #113: High Accelerometer bias estimates and slow convergence on velocity (and occasional divergence)

nplayle, 2015-05-13:
Hi Everyone, 

Apologies for what is hopefully a simple resolution but I've been working on this for a few days now and nothing seems to be working, so I'm hoping someone here can help. 

So my setup consists of an IMU streaming data (in NED) to the computer. I then transform the data from NED to ENU in a ROS serial node and send it to the MSF node. 

PTAM is running on a downward facing camera on the front of the aircraft, and is streaming data on the /vslam/pose topic to MSF. 

The IMU is streaming data that looks as follows:
[image]
Which seems reasonable to me. Z axis is positive up and reading ~9.8m/s. X points out over right wing and increases when you roll the aircraft left. Y points out the nose and increases when you pitch up. X rate is positive when pitching up, Y rate is positive when rolling right. Z rate is positive when yawing left. 

[image]
PTAM is running on the aircraft and is outputting as follows. After initialisation, the orientation of PTAM is ~`[-0.02 -0.99 -0.02 -0.02]` (in wxyz form). As best I can tell, this implies the initial orientation of the camera is rotated 180deg around the IMU x axis, and I initialise q_ic as follows:
`pose_sensor/init/q_ic/w: 0.0`
`pose_sensor/init/q_ic/x: 1`
`pose_sensor/init/q_ic/y: 0`
`pose_sensor/init/q_ic/z: 0`

The camera faces down, and moving the aircraft forward brings in new scenery from the top and old out the bottom. In the image below the top of the image is forward, right of the image is right hand side of aircraft, etc. 
[image]

Now the issue I have is that often times when I initialise MSF it will diverge quite rapidly. When it does remain stable there is a very large bias estimated for the accelerometers and a large bias on velocity (even though the aircraft remains stationary), as shown below. Immediately after initialisation there may be up to a 0.5m/s bias on velocity that doesnt go away. 
Velocity, note in this scenario I forced it to 'converge' to roughly 0m/s by increasing the accelerometer bias noise.: 
[image]
And accelerometer bias: note 1m/s bias. It ranges between 1-4m/s for a near zero velocity estimate. The bias changes largely dependent on movement / initialisation / re-initialisation and doesnt seem to converge to a constant value. 
[image]

I've tried increasing the bias noise which seems to help but makes the bias overly large (and probably breaks other parts of the filter). I've tried exciting the sensors as best I can (UAV is on a cart so only yaw is excited for angular rates). Do you think this is simply a convergence problem or is it a setup issue? I did have it working much better than this a few weeks ago but then merged a more recent version of code which seems to have broke it. I tried reverting back but the issue seems to be there now as well, so I'm wondering if it was there all along?

I'm still trying to work out why the filter spikes seemingly randomly. It may be due to my IMU dropping packets but I don't see how that would cause the spike. 

On a somewhat related note, what would be the best way to initialise the filter with known values? e.g. readings from a compass and ultrasonic altimeter? 

Any help wouldbe greatly appreciated. 
Thanks, 
Nick

simonlynen, 2015-05-13:
@nplayle very likely your camera to imu calibration is not correct. This needs to be set s.t. it denotes the orientation of the camera frame of reference w.r.t. the imu frame of reference and has nothing to do with what PTAM outputs. So you have the z-axis of the camera pointing along the optical axis, x right, y down. Please set the q_ic rotation to the right quaternion that corresponds do your setup.

Everything else that you see is resulting from this calibration being off. Once this is fixed none of the other things you tried are needed, so it is best to make also sure that any changes to the code are reverted.

Simon

simonlynen, 2015-05-13:
Also is your ptam map initialized s.t. the ptam z-axis (the axis orthogonal to the grid) parallel to the direction of gravity? If this is not the case you will need to initialize the q_wv rotation appropriately.

simonlynen, 2015-05-13:
@nplayle also thanks for the detailed description and plots. Great.

nplayle, 2015-05-13:
@simonlynen Thanks for the quick response. 

I was reasonably sure that q_ic is correct but I attached a picture of the setup, maybe you can double check too? Normally we have X pointing out the nose, Y right and Z down but I changed it to correspond to ENU, would this cause an issue? I'm using wxyz = `[0 1 0 0]` for my q_ic in this case which should correspond to a 180deg rotation around the IMU X axis (if I'm correct). I've also set p_ic to correspond to the camera location of 0.81m in positive Y displacement from the IMU. 

[image]

Attached is the report that MSF gives after initialisation. Since it is close to unity I figured that meant the q_ic is correct. 

```
[ INFO] [1431547076.861221521]: initial measurement pos:[-0.191628
0.192392 0.788744] orientation: [-0.0777, -0.994, -0.0443, -0.0693] [ WARN] [1431547076.966630822]: Using simulated core plus fixed diag initial error state covariance.
[ INFO] [1431547076.976773262]: Initializing msf_core (built: May 13 2015) [ INFO] [1431547076.976843640]: Core parameters:
         fixed_bias:     0
         fuzzythres:     0.1
         noise_acc:      0.04
         noise_accbias:  0.004
         noise_gyr:      0.0007
         noise_gyrbias:  3e-06

[ INFO] [1431547076.976965123]: Core init with state:
--------- State at time 7076.87s: ---------
0 : [0-2]        : Matrix<3, 1>         : [-0.139892 -0.602618  0.998699]
1 : [3-5]        : Matrix<3, 1>         : [0 0 0]
2 : [6-9]        : Quaternion (w,x,y,z) : [0.994, -0.0777, -0.0693, 0.0443]
3 : [10-12]      : Matrix<3, 1>         : [0 0 0]
4 : [13-15]      : Matrix<3, 1>         : [0 0 0]
5 : [16-16]      : Matrix<1, 1>         : [1]
6 : [17-20]      : Quaternion (w,x,y,z) : [1, 0, 0, 0]
7 : [21-23]      : Matrix<3, 1>         : [0 0 0]
8 : [24-27]      : Quaternion (w,x,y,z) : [0, -1, 0, 0]
9 : [28-30]      : Matrix<3, 1>         : [    0  0.82 -0.08]
-------------------------------------------------------
```

simonlynen, 2015-05-13:
Can you double check that the IMU orientation you have drawn is exactly the data that also arrives in the filter? (just add some print outs in the IMU callback in the imu-sensor-handler).

simonlynen, 2015-05-13:
> On a somewhat related note, what would be the best way to initialise the filter with known values? e.g. readings from a compass and ultrasonic altimeter?

Altimeter init via the "set_height" service call
Compass init via the "yaw_init". I am not sure if this is still exposed, but you can add this in the filter-init call.

nplayle, 2015-05-13:
I added some ROS_INFO statements in the IMUCallback() function and it shows the same values as what's being sent.

It does eventually converge to some reasonable value but just starts off high, and motion doesn't seem to affect the convergence rate until it has been operating for some time. Do you need to have extended excitation to have it work properly? It seems like the initial bias is being estimated incorrectly which is causing the velocity errors, but I don't know why the initial bias estimate jumps up so high (pictures coming).

simonlynen, 2015-05-13:
Sorry for asking these questions, but did you make sure that you get the acceleration measurement corresponding to the gravitational force in the directions that correspond to your drawing?

No, you should not need any excitation and you shouldn't see jitter/jumps at all. What you have in the plots is definitely neither normal nor correct.

nplayle, 2015-05-13:
This is what I see when I initialise the filter (at 7.5), leave it stationary for approx 10mins and then move the uav around (at approx 8.04) 
[image]
And velocity:
[image]

Obviously movement is better for convergence, but I'm wondering what is causing the initial bias estimate to shoot up so high? The uav is not moved until around t=8.04 so I can't see a reason why it would estimate them so high.

nplayle, 2015-05-13:
>did you make sure that you get the acceleration measurement corresponding to the gravitational force in the directions that correspond to your drawing?

I double checked and am reasonably sure that is correct.  When I tilt the aircraft left side down I get a positive X reading and pitch up is a positive Y reading. Rolling right is positive Y rate, pitching up is positive X rate.

The PTAM frame is set so that the grid is parallel with the floor (i.e. z axis points vertically up and down). 

I'll try plotting q_wv and see if those values make sense.

>you shouldn't see jitter/jumps at all
I think this might be coming from my IMU. I'll have a closer look at it, I'm wondering if my parsing / NED->ENU script is causing issues occasionally. 

Nick

nplayle, 2015-05-14:
I think I've got this solved now (at least until I start moving around again). It was a combination of two problems:
1. Incorrect camera calibration meant that the effective q_wv was chanigng any time I moved the UAV. I think they got overwritten when I did a git stash and checked out an old version. 
2. The autopilot I'm using seems to have a suspiciously high gyro bias (0.02rad/s, 0.01 rad/s). With the default parameters the gyro bias wasn't being estimated fast enough which led to the incorrect velocity estimates. I set the gyro bias noise higher (0.001) and it seems to converge on the bias much faster. I suspect I could then lower the bias noise back down but rqt won't let me do that (I have a choice between 0.001 and 0, so I'll give that a change). 

I suppose I could add the initial bias estimates to the filter but it would be kind of hacky and susceptible to change. 

Thanks for the help @simonlynen, greatly appreciated. 

Nick

closed 2015-05-14 by nplayle

simonlynen, 2015-05-14:
@nplayle thanks for the feedback. Happy to hear you found the sources of the issues.

Storing an initial linearization point for the bias is usually a good idea if you find your IMU having roughly consistent values from turn-on to turn-on. In the filter-init method you could read these from the yaml and store them back when you shutdown the filter (given that they have remained bounded and the filter is in a stable state)