ethz-asl/ethzasl_msf #114: Estimation Jitter Causing Instability

nplayle, 2015-05-21:
Hi Again, 

As my plots in #113 showed I was seeing spikes in the state estimate for no apparent reason. 

[image]
[image]

I've done some more digging and I still can't find the reason for them. I tried a second IMU (first consisted of ADXRS642 and ADIS16006, now an MPU6000) to no avail. The rqt_plot for the IMU shows smooth readings with no jumps. Similarly, the rqt_plot for the ptam vslam/pose topic also shows no problems, everything is smooth. The IMU is updating at ~83Hz (former IMU was 90Hz), and the visual slam at 18Hz according to rostopic hz. 

Somewhere along the line it is being corrupted and directly influencing the state estimates. Sometimes, the spike is temporary and looks to be for one pose update. Other times it seems as though the spike incfluences the estimate and actually causes the filter to go unstable (velocities spike to 50 000+, position also spikes up). Occasionally it seems to converge on a value that is not correct (like 500m, 300m, 100m for states 0,1,2). This happens without any external sensor input (i.e. the UAV is sitting there untouched). 

Interestingly enough this behaviour occurs both in SSF (ethzasl_sensor_fusion) and MSF, so I'm leaning toward it being a problem with my setup rather than one with the filter itself (that and it used to not have all these spikes). But all the debugging I've done shows the values getting to the sensor as anticipated. 

The covariance values are the defaults. 

I'll update with a more recent picture shortly.

nplayle, 2015-05-21:
Here you can see the estimates from PTAM remaining sane but the MSF estimates diverging. This was after a ~20cm translation in the X axis ('east')

[image]

The startup report shows:
```
process[msf_pose_sensor-1]: started with pid [23179]
[ INFO] [1432237038.949669866]: /msf_pose_sensor:
	subscribed to topics:
	advertised topics:
		/rosout
		/tf
		/msf_pose_sensor/core/parameter_descriptions
		/msf_pose_sensor/core/parameter_updates
		/msf_core/state_out
		/msf_core/correction
		/msf_core/pose
		/msf_core/odometry
		/msf_core/pose_after_update
		/msf_core/ext_state
		/msf_core/cov_core
		/msf_core/cov_aux
		/msf_core/cov_core_aux

[ INFO] [1432237038.956100618]: Loading parameters for pose sensor from namespace: /msf_pose_sensor/pose_sensor
[ INFO] [1432237038.958467503]: Pose sensor is interpreting measurement as world w.r.t. sensor (e.g. ethzasl_ptam)
[ INFO] [1432237038.958493564]: Pose sensor is using covariance from sensor
[ INFO] [1432237038.958504506]: Pose sensor is handling measurements as absolute values
[ INFO] [1432237039.209917293]: *** pose sensor got first measurement from topic //vslam/pose ***
[ INFO] [1432237062.040215579]: initial measurement pos:[ -0.3719 0.184747  1.57706] orientation: [-0.0101, -0.999, 0.0219, -0.0221]
[ WARN] [1432237062.145636990]: Using simulated core plus fixed diag initial error state covariance.
[ INFO] [1432237062.155764819]: Initializing msf_core (built: May 19 2015)
[ INFO] [1432237062.155810660]: Core parameters: 
	fixed_bias:	0
	fuzzythres:	0.1
	noise_acc:	0.0022563
	noise_accbias:	5e-08
	noise_gyr:	0.0004
	noise_gyrbias:	3e-05

[ INFO] [1432237062.155868112]: Core init with state: 
--------- State at time 7062.05s: ---------
0 : [0-2]	 : Matrix<3, 1>         : [-0.410795 -0.612623   1.67239]
1 : [3-5]	 : Matrix<3, 1>         : [0 0 0]
2 : [6-9]	 : Quaternion (w,x,y,z) : [0.999, -0.0101, -0.0221, -0.0219]
3 : [10-12]	 : Matrix<3, 1>         : [0 0 0]
4 : [13-15]	 : Matrix<3, 1>         : [0 0 0]
5 : [16-16]	 : Matrix<1, 1>         : [1]
6 : [17-20]	 : Quaternion (w,x,y,z) : [1, 0, 0, 0]
7 : [21-23]	 : Matrix<3, 1>         : [0 0 0]
8 : [24-27]	 : Quaternion (w,x,y,z) : [0, -1, 0, 0]
9 : [28-30]	 : Matrix<3, 1>         : [    0   0.8 -0.08]
-------------------------------------------------------
[ WARN] [1432237094.039439077]: Fuzzy tracking triggered: 0.182509 limit: 0.1
```
(more fuzzy tracking warnings follow). 
Does having the ```Using simulated core plus fixed diag initial error state covariance.
``` warning cause any issues?

MarcGyongyosi, 2015-05-21:
What's the rotation of your camera w.r.t to the IMU? I.e. What's its position on the MAV?

----------------------------------------
Marc Gyongyosi
----------------------------------------

nplayle, 2015-05-21:
It's the same as seen in #113:

The camera is on sticks so the lens doesn't see the cart the UAV is on. 

[image]

When looking at the image from the camera the top of the image is forward, the bottom of the image is towards the bcak of the UAV, and right on the image is right on the UAV. Based on that I came up with the frame in the image, I'm reasonably sure it's correct but I might be wrong.

simonlynen, 2015-05-21:
If you get immediately a fuzzy tracking then there can only be three
reasons. Wrong camera to IMU transformation, timestamps of IMU and camera
are not on the same basis / clock or your ptam pose gravity direction does
not correspond to the world gravity. Since we have already clarified the
ptam pose gravity question I would suggest you check again the other two
items.

nplayle, 2015-05-21:
The fuzzy tracking is not necessarily immediate though. Occasionally it works fine and is trackable for a few minutes with the spikes in the states. You can even move the setup around without issue. 

It's just horribly unreliable and the spikes in the states occasionally cause fuzzy tracking and divergence. In that last posting there was 30s between filter initialisation and fuzzy tracking being triggered. 

I tried to verify the times and they look like they're in sync. The IMU node is using rospy though so is it posisble it has a slightly different time basis than PTAM? 

The other IMU transforms i've tried all seem to cause much faster divergence than the one I currently use.

nplayle, 2015-05-22:
Well I've narrowed this down to a specific computer problem. 
I cloned my repo to my laptop and compiled there, same version of ROS and ubuntu. 
Running on my laptop gets ~21Hz vslam rate @ ~180hz imu rate, no jitter. 

Will probably reformat the on board computer and see if that fixes the problem. (Running on a gigabyte brix)
Closing for now, will report back later on what the problem was (or if it was solved by reformat). 

Thanks again for your help.

closed 2015-05-22 by nplayle