ethz-asl/ethzasl_msf #119: clarification needed re. orientation of IMU and pose sensor frames

MarcGyongyosi, 2015-07-30:
Hi,

I am back again and I hope maybe this time I could clear this up for once and all: We are trying to benchmark this sensor fusion framework against our own for evaluation purposes but I still can't get it to work properly. At this point I am very sure this is most likely because of the rotation parameters set in the configuration file for the pose sensor. Here's a quick and hopefully understandable overview:

We have a Pixhawk Flightcontroller onboard whose IMU output we want to fuse with the output of a forward facing camera running ORB-SLAM onboard. 

The IMU on the pixhawk is oriented such that (looking at the MAV) Z+ is up, Y+ is left, and X+ is forward. 

The camera is pointing forward at the front of the MAV. It is providing PoseStamped output about the camera's position w.r.t. to its point of initialization. If I move the camera I get the following coordinate system orientation: Z+ forward, X+ right, Y+ down. 

So looking at camera and IMU we see that:

IMU -- ORB-SLAM
X         Z
Y         -X
Z         -Y

Given that we are supposed to provide q_ic to identify the rotation of the camera to the IMU in the IMU frame this gives a rotation quaternion q_ic of:
x = 0.5
y = -0.5
z = 0.5
w = -0.5

Previously, in #102, #91, and #83 it was suggested that for such a setup, the proper rotation quaternion would be:

x = 0.5
y = -0.5
z = 0.5
w = 0.5

I visualized this in RVIZ and I am fairly certain that the latter is wrong and that it has to be the first of the two quaternions. Also, with the second quaternion I am not able to get the pose estimate to converge upon initialization. 

I am able to observe how the pose estimate converges on a "proper" value when the MAV is at rest and I use the first q_ic settings. 

So compared to all previous issues I opened on here, I seem to have gotten further than before. 

Now comes the tricky part: while the pose estimate is somewhat there, I can see it spike and jitter a lot, even when the MAV is at rest. Also, the moment I move it, the wrong axes start changing, i.e. if I move it up, the y axes seems to change on the pose output - it should be the z-axis, right? 

In addition, the pose estimates are not accurate at all, they seem to vary unpredictably and are not "to-scale"... 

I really want to compare our simplistic approach with this framework as I think that would be a very valuable test to see how well our approach is doing compared to MSF, however, I really can't get this to work here... 

I hope somebody can help me, 

specifically, I need help verifying my q_ic values and some possible explanations for why - even though the pose estimates seem to converge - I get wrong results upon moving the MAV. 

On a side note: my camera is calibrated....  

Thanks a lot!

Marc

MarcGyongyosi, 2015-08-03:
Alright, so I am getting good results now (finally!). What I did is rotate the pose measurement in a separate node directly and then feed that manually rotated measurement into the EKF.

I am surprised it didn't work when I had it set up in ethzasl_msf with the q_ic value (the first one above) since that's exactly the rotation I am applying now when I rotate externally. 

Anyways, for anyone who runs into this issue: from what I can see, all the "rotation of pose sensor into IMU frame in IMU frame" means is that you have to make the measurements of the pose sensor align with the IMU frame. I.e. if IMU's x is forward then the output of the pose sensor in that direction should also be x. Same with all other axes and roll pitch yaw. Once that is set up and the scale of your VSLAM framework is not changing drastically very quickly, you're golden and all set up!

After all, thanks for this great piece of software, now it finally works, even with a forward facing camera!

Marc

closed 2015-08-03 by MarcGyongyosi

simonlynen, 2015-08-03:
Hi Marc,

Thanks for the feedback. Sorry this didnt work out right away. Ich guess there might be different conceptions about how the rotation is applied which might have caused the issue here.

Simon

AlexandreBorowczyk, 2016-01-15:
Hi Marc,

I was wondering if you ever managed to get this work with the config file. I have a very similar setup, a front facing PointGrey camera with a Pixhawk and as for you the I.C. transform doesn't affect the outcome of the filter.

Thanks
Alex

MarcGyongyosi, 2016-01-15:
I was never able to fix it by setting the parameters in the config file. I
always rotate externally.

best,
Marc

AlexandreBorowczyk, 2016-01-17:
Alright! Thanks I will let you know if I get something working!
Cheers

AlexandreBorowczyk, 2016-01-26:
Added the orientation correction directly in ORB_Slam. Works fine!
@MarcGyongyosi: what performance in terms of estimation quality have you been able to obtain from the ORB_SLAM-MSF combo? I'm not sure how to interpret my results.
Thanks

bishwa9, 2016-07-06:
@MarcGyongyosi , @simonlynen : I too am struggling to get the MSF framework up and running with ORB_SLAM. For me, the frames are alligned as such:

IMU ORB_SLAM
x -x
y z
z y

As such, the Rotation matrix from slam to imu frame (the matrix used to rotate points in slam frame ot imu frame) is 
-1 0 0
0 0 1
0 1 0

From this I get the q_ic:
w = 0 
x = 0
y = 0.707107
z = 0.707107

I use your way and use the above rotation matrix to rotate ORB_SLAM's ouptut to imu frame, I invert it before this so I have sensor in world frame. I feed this in, however, I have to provide the above q_ic in the settings file.

Using this, I get it to sit stably. However, much movement makes the system unstable (fuzzy tracking followed with scale trying to go negative). Also, without movement, the pose outputs aren't scaled appropriately.

Thomas00010111, 2016-08-05:
Hi,

Do not know if this thread is still alive... 
I am using a small Intel RGBD (I think it is called DS4 but not sure) camera together with orb slam 2 and a pixhawk. I am using the inav module which accepts vision input. I added some code to the orbslam2 rgbd example to get the correct rotation. When checking the /mavros/local_setpoint/pose topic the coordinates look ok and change like expected. (x: forward, y:right, z:up).
Why did you chose the MSF frame work instead of the existing filters/modules from the px4 code base? Do you explicitly handle delays between camera and IMU? How are the experiences so far?

Cheers