ethz-asl/ethzasl_msf #120: Best practices for initialization

MarcGyongyosi, 2015-08-05:
Hi,

I got the framework to work properly with ORB-SLAM onboard our MAV, now I would like to know what you suggest in terms of best practices for initialization. From what I can see, currently we have the plain init_filter and init_height options available. Since I can only roughly determine my height by hand/eye while holding it in my hand, I found the init_filter on ground to be the most reliable option (I initialize my VSLAM framework, then I put the MAV to rest on the ground at its origin, then I hit init_filter, then I lift it up and shake the MAV before putting it back down). 

There's only one problem, MSF only sometimes initializes to the correct xyz coordinates when the MAV is on the ground (i.e. sometimes it's nicely at 0,0,0, but sometimes it initializes at arbitrary points (e.g. 0.1,0.05,-0.2) I was wondering if there is a way to call something like "initialize at origin" which sets all the initial values to 0 (or close to 0). That way we could consistently keep our local origin of our MAV frame positioned within the physical world and we could also align it with a global map.

Maybe someone can shed some light on this, or point me to the right resources to implement this on my own.

Thanks a lot!

Marc

simonlynen, 2015-08-05:
@MarcGyongyosi The initial position is taken from the SLAM system output. If ORB-SLAM returns you a position which is non-zero for the initial pose, then this will be used also by the estimator. What you can do is to push this offset into the vision-world transformation during the init of the estimator, then the estimator will return [0, 0, 0] independent of the initial offset of ORB-SLAM:
https://github.com/ethz-asl/ethzasl_msf/blob/b5102f40ff313e1524cbb9287294d28a3a45f7d1/msf_updates/src/position_pose_msf/position_pose_sensormanager.h#L139

MarcGyongyosi, 2015-08-05:
@simonlynen understood, thanks for getting back so fast. I was already wondering about this offset in general when using init_height since an offset in the pose sensor (e.g. from moving the camera up in the z direction to get the initial map) would probably result in improperly calculated scale when calling init_height (looking at https://github.com/ethz-asl/ethzasl_msf/blob/master/msf_updates/src/pose_msf/pose_sensormanager.h#L140) if I'm not wrong?  thanks!

simonlynen, 2015-08-05:
The scale is calculated as the ratio between the height reported by the
slam and the height given in dynamic reconfigure. this is not influenced by
what I had written in the last mail.

MarcGyongyosi, 2015-08-05:
That's interesting so let's say vslam reports a height of -0.1m on the ground and it initialized 0.25m (real world) above surface. So the vslam scale is approximately .1/.25 (=.4) and the VSLAM origin is offset in z by .1 (vslam units). Then I call init_height at .2m (real world) at which point vslam might report something like -0.02. The scale calculated in the line I pointed to before would now be -0.02/0.2 (i.e. p[2]/req.height) = -0.1. Doesn't a negative scale always get automatically corrected to 0.1? I remember seeing warning messages about that. In order to get the proper initial scale, wouldn't you have to subtract the offset from initialization right away? So for this case to get the right initial scale estimate, wouldn't you have to add 0.1 (the initialization offset of vslam) to the p[2] measurement before dividing by the provided height estimate? That way you would get .08/.2 ( (p[2]+z_offset)/req.height) which is the same (.4) as we determined by inspection before and thus the correct initial scale. 

I am by no means an expert in this and if I'm wrong feel free to just tell me and you don't have to explain further - I don't want to waste your time, I wouldn't be offended. I just want to make sure I understand everything properly! Thanks so much!

simonlynen, 2015-08-05:
I dont see how your vslam can report a negative height above the ground.
Since the slam position should denote the center of the camera wrt. the
floor.

MarcGyongyosi, 2015-08-05:
Oh, that's because the camera is facing forward. I probably should have mentioned that...

simonlynen, 2015-08-05:
Still how can you have that below the floor? :)

MarcGyongyosi, 2015-08-05:
If you move the camera up and it initializes at a point above the floor, it sets its origin where it initializes. From then on its position is relative to that first point of initialization. So as I described earlier, if you move up the camera (in z direction) by 30 cm and it only initializes in the last 5cm, then your origin for the camera frame is shifted up. And then if you go down from that first point of initialization, then you have "negative" height readings because you're below your vslam origin. Anyways, I guess this is a special case since the camera is mounted forward looking and I am using ORB-SLAM. ORB-SLAM takes a while to initialize, other than LSD-SLAM and SVO, however it seems to be more robust than those. Using the initial position estimate as an offset seems to be working for me.

MarcGyongyosi, 2015-08-05:
Also, @simonlynen I really appreciate your help/input. Thank you very much!

simonlynen, 2015-08-05:
Ok as mentioned before the two inputs need to refer to the same frame of
reference. If you set your origin for the slam 30cm up then you should also
count the height for the init from that point. I would just use the floor
for both.

closed 2015-08-05 by MarcGyongyosi