ethz-asl/ethzasl_msf #121: Is it possible to use camera pose with unknown scale and GPS readings?

saihv, 2015-08-06:
Hi, I am working on pose estimation of one vehicle with respect to another, where both vehicles have a camera and a GPS module. The translation data coming from the camera algorithm is up-to-scale, but as the extrinsics are changing, the scale is unknown, so it's basically just a unit vector giving me the direction of the translation. But the GPS does give me relative position, although it has its own noise parameters. Is it possible to use this sensor fusion framework with both these kinds of data to obtain a filtered metric translation?

simonlynen, 2015-08-07:
@saihv yes you can use these inputs and adapt the estimator for your setup. I would suggest you first get familiar with the framework by combining a single vehicle with gps and later add the second

saihv, 2015-08-08:
Thank you. Just a follow-up: Is providing IMU data mandatory? Or can I replace that with the GPS and use the camera based data as the "pose sensor"?

simonlynen, 2015-08-08:
This is an indirect kalman filter so you need IMU data at the moment. It
would be possible to replace the IMU propagation with a motion model but we
havent planned to implement this at the moment.

tdnet12434, 2016-04-01:
@simonlynen 
I have problem when visual data gone, The KF can't observe the real yaw angle.
so if i have ahrs data (imu->orientation) for feed into IMUmsghandler , It will fix rotational drift issue?