ethz-asl/ethzasl_msf #122: Missing Dataset

goldbattle, 2015-08-10:
I was wondering is the dataset mentioned in [this tutorial](http://wiki.ros.org/ethzasl_sensor_fusion/Tutorials/Introductory%20Tutorial%20for%20Multi-Sensor%20Fusion%20Framework) is still avalible for download someplace.
The download [link](http://wiki.ros.org/ethzasl_sensor_fusion/Tutorials/Introductory%20Tutorial%20for%20Multi-Sensor%20Fusion%20Framework?action=AttachFile&do=upload_form&ticket=0055c8c4e8.c1f03301a779c116c3de56d7e267a067583b424a&target=dataset.bag) currently takes me to a list of the images attached to the article, but not a `.bag` file.

Thanks!

goldbattle, 2015-08-10:
Seems that the advance tutorial [here](http://wiki.ros.org/ethzasl_sensor_fusion/Tutorials/custom_sensor_design) links to it correctly.

v0n0, 2015-12-15:
I was also wondering if any dataset with groundtruth is available for testing my implementation of msf.