ethz-asl/ethzasl_msf #124: No output on /msf_core/state_out with ORB SLAM

ashariati, 2015-08-13:
Hello!

So I've gotten through the tutorial with the viconpos_sensor example just fine without issue. However, now I would like to use the output from ORB SLAM [http://webdiis.unizar.es/~raulmur/orbslam/] as my pose_sensor. My only modification to their software has been to publish their /ORB_SLAM/World to /ORB_SLAM/Camera transform as a geometry_msgs::TransformStamped in following their broadcast of the transform on /tf. The topic name is /ORB_SLAM/camera_pose.

Below is my configuration:
```
/pose_sensor/core/data_playback: true
/pose_sensor/core/core_fixed_bias: false

/pose_sensor/core/core_noise_acc: 0.083
/pose_sensor/core/core_noise_accbias: 0.0083
/pose_sensor/core/core_noise_gyr: 0.0013
/pose_sensor/core/core_noise_gyrbias: 0.00013

/orbslam_pose_sensor/pose_sensor/pose_noise_scale: 0.0     
/orbslam_pose_sensor/pose_sensor/pose_noise_p_wv: 0.0                  
/orbslam_pose_sensor/pose_sensor/pose_noise_q_wv: 0.0                  
/orbslam_pose_sensor/pose_sensor/pose_noise_p_ic: 0.0
/orbslam_pose_sensor/pose_sensor/pose_noise_q_ic: 0.0
/orbslam_pose_sensor/pose_sensor/pose_delay: 0.003
/orbslam_pose_sensor/pose_sensor/pose_noise_meas_p: 0.01    # [m]
/orbslam_pose_sensor/pose_sensor/pose_noise_meas_q: 0.02    # [s]
/orbslam_pose_sensor/pose_sensor/pose_initial_scale: .166

/orbslam_pose_sensor/pose_sensor/init/q_ic/w: 1.0
/orbslam_pose_sensor/pose_sensor/init/q_ic/x: 0.0
/orbslam_pose_sensor/pose_sensor/init/q_ic/y: 0.0
/orbslam_pose_sensor/pose_sensor/init/q_ic/z: 0.0
/orbslam_pose_sensor/pose_sensor/init/p_ic/x: 0.0   
/orbslam_pose_sensor/pose_sensor/init/p_ic/y: 0.0
/orbslam_pose_sensor/pose_sensor/init/p_ic/z: 0.0

# Transformation that expresses the position and orientation of the gravity aligned world frame 
# w.r.t the vision/camera frame
/msf_viconpos_sensor/pose_sensor/pose_fixed_scale: true
/orbslam_pose_sensor/pose_sensor/pose_fixed_p_wv: true              
/orbslam_pose_sensor/pose_sensor/pose_fixed_q_wv: true

# Transformation that expresses the position and orientation of the pose-sensor w.r.t. the IMU 
# frame of reference, expressed in the IMU frame of reference.
/orbslam_pose_sensor/pose_sensor/pose_fixed_p_ic: true
/orbslam_pose_sensor/pose_sensor/pose_fixed_q_ic: true

/orbslam_pose_sensor/pose_sensor/pose_absolute_measurements: true
/orbslam_pose_sensor/pose_sensor/pose_use_fixed_covariance: true
/orbslam_pose_sensor/pose_sensor/pose_measurement_world_sensor: true  
```

And here is my launch file:
```
<launch>

    <node  name="orbslam_pose_sensor" pkg="msf_updates" type="pose_sensor" clear_params="true" output="screen"> 

        <remap from="msf_core/imu_state_input" to="quadcloud2/kmel_kquad_interface/imu" />
        <remap from="msf_updates/transform_input" to="ORB_SLAM/camera_pose" />

        <rosparam file="$(find msf_updates)/orbslam_sensor_fix.yaml"/>
    </node>

</launch>
```

I initialize the filter with rqt_reconfigure.

I am seeing data on the /msf_core/pose topic, however when I plot the data, it just looks like a long drift on the y-axis.

Thanks in advance

MarcGyongyosi, 2015-08-13:
Hi, have you checked that the vision frame is aligned with the IMU frame?

ashariati, 2015-08-14:
So, if we are considering naming convention in the graphic under Section 4 : General Setup on the tutorial below
 
http://wiki.ros.org/ethzasl_sensor_fusion/Tutorials/Introductory%20Tutorial%20for%20Multi-Sensor%20Fusion%20Framework

I have made sure that IMU frame and the Camera frame are aligned, yes

MarcGyongyosi, 2015-08-14:
Just to make sure, because there is often a lot of confusion with this:
When you move your quad in the x-direction of your IMU, then that's also
the x-direction of the ORB-SLAM output? Same for the other axes? Basically
what the graph means, is that the x,y,z axes of the output of the
"blackbox" VSLAM framework need to align with the IMU measurements after
rotating and translating by q_ic and p_ic respectively. However, I have
found mixed results with using the q_ic and p_ic values even if (at least
from my understanding) they should have been correct. That's why we rotate
the VSLAM measurement externally in a separate node.

ashariati, 2015-08-14:
Hmm. I am now less sure that my IMU and Camera frame are aligned...

Unfortunately, I am working with logged data that I did not collect myself (the person that collected the data and manages the robots is on vacation), so I can't be sure. Of course, once he gets back I'll be sure to ask and make sure.

I had setup a little experiment to see if the frames were infact aligned by republishing data from the IMU as a transformation between the IMU frame and the /ORB_SLAM/World frame (or with this terminology the vision frame). Since I do not have translation from the IMU, I used the translation of /ORB_SLAM/Camera (I suspect this could have been my wrong step). When I visualized these transforms together, it appeared that the IMU frame and the Camera frame were aligned. However, I am not very confident in this process anymore.

All the same though. I guess the question I have is: Even if this transformation is wrong, should there still be no data being published on the /msf_core/state_out topic? I could understand the output being incorrect if my transformation between IMU and Camera was incorrect, but is this in fact the correct behaviour to give nothing? Therefore, should I expect that once I fixed this transformation, I should just start getting output?

MarcGyongyosi, 2015-08-14:
Yeah the fact that you are not getting any data on the state topic is very weird. If you want, send me an email (via http://ifm-tech.com so I don't have to post my email here) and I can send you a rosbag from our setup for you to test. How is the quad setup in your experiments? We have a forward facing camera running ORB-SLAM in standard configuration; IMU is oriented such that x is forward, y is left, z is up. Then you can run that bag file through your setup and see if it still doesn't work. I must also say that I haven't worked with tf input into msf, maybe somebody else can give some info on that? At some point, I added a pose publisher in ORB-SLAM which I've been using since (just in the Tracking file). On our quad it works really well....

ashariati, 2015-08-14:
Sure, that would be awesome.

So the quad is setup with a downward facing camera, running ORB-SLAM in the standard configuration as well. There is an IMU and GPS on board, however, I cannot speak to the orientation of the IMU since I do not even know which quadrotor my colleague used to record the data, but I'll be sure to ask once he is back at the lab.

Also, I am not using tf as an input to msf either. I did the same thing as you, I added a pose publisher as well in the Tracking file. However, I publish that data as a geometry_msgs::TransformStamped, and use this as the input to msf.

I  suspect I am doing something silly....

ashariati, 2015-08-18:
Just wanted to post the solution to this problem here in case anyone deals with this in the future.

If anyone faces this problem, consider checking the frequency at which your imu publishes data.

I discovered that my IMU was publishing data at a rate of 240 Hz, which caused the node to be unable to finish computing the filter update before the arrival of the next message. I downsampled the IMU messages to 80 Hz and am now getting output.

closed 2015-08-18 by ashariati

bishwa9, 2016-07-06:
@ashariati 

Hello,

I am running into similar issues as you guys. I am trying to run ORB with the framework. Please help.

My frames:

The imu-orb orientation for your help:

IMU       ORB
x            -x
y            z
z            y

The Rotation matrix from imu to orb is
-1 0 0
0 0 1
0 1 0

The quaternion
/msf_viconpos_sensor/pose_sensor/init/q_ic/w: 0
/msf_viconpos_sensor/pose_sensor/init/q_ic/x: 0.0
/msf_viconpos_sensor/pose_sensor/init/q_ic/y: 0.707107
/msf_viconpos_sensor/pose_sensor/init/q_ic/z: 0.707107

Does this make sense?

I am also applying the Rotation matrix above to rotate the ORB_SLAM output (in world) to the imu frame, just like @ashariati  does in his ORB_SLAM repository.

I get a stable initialization but upon movement it is very jittery and the scale very easily drops to 0. 

This is my settings file
[image]