ethz-asl/ethzasl_msf #127: Measurement units and orientations?

boeboe26, 2015-11-03:
Hi!

We have a system with a 100Hz IMU (Gyro+Accelerometer) and a 10Hz Camera onboard. 

We conducted two experiments to verify the functionality of the framework. Unfortunately, there seems to be a problem with the IMU that we could not resolve.

Could there be something wrong with the measurement units or the orientation of the axes?
Your help would be greatly appreciated!

All plots are identically color coded: 
x-axis: blue 
y-axis: red
z-axis: green

Here is the expected output, calculated separately with the registration output. The 'x' indicate an input pose measurement. You can clearly see a gap in the registrations at about 40s:
[image]

Here are the estimated positions (output of the framework):
[image]

Looks great, doesn't? Unfortunately, the propagated positions in between the registrations is not reasonable and drifts greatly. You can clearly see the drift in the pose measurement gap:
[image]

It looks like the whole system is accelerating towards some strange gravity? Is the accelerometer input ok?

We adopted the vicon example with a viconpos_sensor_fix.yaml that looks like this: 
We set the core_noise_accbias to 1/10th of core_noise_acc and
core_noise_gyrbias as 1/10th to core_noise_gyr
Is this correct?

    /msf_viconpos_sensor/core/data_playback: true
    /msf_viconpos_sensor/core/fixed_bias: 0
    /msf_viconpos_sensor/core/core_noise_acc: 0.0002452 #[m/s^2/sqrt(Hz)] 
    /msf_viconpos_sensor/core/core_noise_accbias: 0.00002452 
    /msf_viconpos_sensor/core/core_noise_gyr: 0.0005236 # [rad/s/sqrt(Hz)] 
    /msf_viconpos_sensor/core/core_noise_gyrbias: 0.00005236 
    /msf_viconpos_sensor/pose_sensor/fixed_scale: 1
    /msf_viconpos_sensor/pose_sensor/pose_noise_scale: 0.0
    /msf_viconpos_sensor/pose_sensor/pose_noise_p_wv: 0.0
    /msf_viconpos_sensor/pose_sensor/pose_noise_q_wv: 0.0
    /msf_viconpos_sensor/pose_sensor/pose_noise_q_ic: 0.0
    /msf_viconpos_sensor/pose_sensor/pose_noise_p_ic: 0.0
    /msf_viconpos_sensor/pose_sensor/pose_delay: 0.01 #[ms]
    /msf_viconpos_sensor/pose_sensor/pose_noise_meas_p: 0.0005 #[m]
    /msf_viconpos_sensor/pose_sensor/pose_noise_meas_q: 0.02 #[rad]
    /msf_viconpos_sensor/pose_sensor/core_height: 0.20
    /msf_viconpos_sensor/pose_sensor/init/q_ic/w: 1.0
    /msf_viconpos_sensor/pose_sensor/init/q_ic/x: 0.0
    /msf_viconpos_sensor/pose_sensor/init/q_ic/y: 0.0
    /msf_viconpos_sensor/pose_sensor/init/q_ic/z: 0.0
    /msf_viconpos_sensor/pose_sensor/init/p_ic/x: 0.04  [m]
    /msf_viconpos_sensor/pose_sensor/init/p_ic/y: -0.04 [m]
    /msf_viconpos_sensor/pose_sensor/init/p_ic/z: -0.280 [m]
    /msf_viconpos_sensor/pose_sensor/pose_fixed_scale: true
    /msf_viconpos_sensor/pose_sensor/pose_fixed_p_ic: true
    /msf_viconpos_sensor/pose_sensor/pose_fixed_q_ic: true
    /msf_viconpos_sensor/pose_sensor/pose_fixed_p_wv: true
    /msf_viconpos_sensor/pose_sensor/pose_fixed_q_wv: true
    /msf_viconpos_sensor/pose_sensor/pose_absolute_measurements: true
    /msf_viconpos_sensor/pose_sensor/pose_use_fixed_covariance: true
    /msf_viconpos_sensor/pose_sensor/pose_measurement_world_sensor: false  # selects if sensor measures its position w.r.t. world (true, e.g. Vicon) or the position of the world coordinate system w.r.t. the sensor (false, e.g. ethzasl_ptam
    /msf_viconpos_sensor/pose_sensor/pose_measurement_minimum_dt: 0.0 #0.05  # Sets the minimum time in seconds between two pose measurements.

Registration output:
[image]
[image]

The Gyroscope IMU delivers very similar results:
[image]

I would estimate the bias to be about 
x-axis: -0.004 rad/s
y-axis: -0.01 rad/s
z-axis: 0.01 rad/s

The framework, however, estimates the gyro biases slightly differently:
[image]

Here is the IMU acceleration, right hand system with the gravity vector aligned to z:
[image]

And the estimated acceleration bias of the framework:
[image]

In a second example, we set 
    /msf_viconpos_sensor/pose_sensor/pose_noise_meas_p: 1 #[m]
    /msf_viconpos_sensor/pose_sensor/pose_noise_meas_q: 1 #[rad]
This should allow the registration algorithm to drift, whereas the IMU should be considered as more confident:

The estimated positions are then:
[image]
The estimated positions are then in the range of meters, instead of centimeters...

v0n0, 2016-01-22:
Did you find out what the problem was?

boeboe26, 2016-01-26:
Yes, through simulations we found out that there was a problem in the hardware... Thanks for asking, I will close the issue.

closed 2016-01-26 by boeboe26