ethz-asl/ethzasl_msf #128: Constant circular motion: Sudden movements

boeboe26, 2015-11-10:
Hi there!

We simulated a very simple, non-linear, motion to be estimated by the framework: a constant circular motion in the x-y plane:

Input trajectory:
[image]

Input acceleration:
[image]

There is no additional rotation, gyroscope input is zero, vision rotation is zero.

The IMU operates at 100Hz, the vision black box at 10Hz. There is no noise in the data, we simulated a perfect IMU using https://github.com/martinling/imusim, a powerful tool!

The framework estimates the following trajectory:
[image]

As you can see, the estimation starts correctly and is very smooth at the beginning. After some time, there is sudden motion at the rate of the updates.

How to explain that?

- Linearization problems of the Kalman filter? How to solve them?
- Nyquist problem? 
- Setting problem?

We used the following settings:
```
/msf_viconpos_sensor/core/data_playback: true
/msf_viconpos_sensor/core/fixed_bias: 0

#########IMU PARAMETERS#######
/msf_viconpos_sensor/core/core_noise_acc: 0.0002452 #[m/s^2/sqrt(Hz)]
/msf_viconpos_sensor/core/core_noise_accbias: 0.00002452 
/msf_viconpos_sensor/core/core_noise_gyr: 0.0005236 # [rad/s/sqrt(Hz)] 
/msf_viconpos_sensor/core/core_noise_gyrbias: 0.00005236 

/msf_viconpos_sensor/pose_sensor/fixed_scale: 1
/msf_viconpos_sensor/pose_sensor/pose_noise_scale: 0.0
/msf_viconpos_sensor/pose_sensor/pose_noise_p_wv: 0.0
/msf_viconpos_sensor/pose_sensor/pose_noise_q_wv: 0.0
/msf_viconpos_sensor/pose_sensor/pose_noise_q_ic: 0.0 
/msf_viconpos_sensor/pose_sensor/pose_noise_p_ic: 0.0 
/msf_viconpos_sensor/pose_sensor/pose_delay: 0
/msf_viconpos_sensor/pose_sensor/pose_noise_meas_p: 0.0005 #m
/msf_viconpos_sensor/pose_sensor/pose_noise_meas_q: 0.02 #rad
/msf_viconpos_sensor/pose_sensor/core_height: 0.20 

/msf_viconpos_sensor/pose_sensor/init/q_ic/w: 1.0
/msf_viconpos_sensor/pose_sensor/init/q_ic/x: 0.0
/msf_viconpos_sensor/pose_sensor/init/q_ic/y: 0.0
/msf_viconpos_sensor/pose_sensor/init/q_ic/z: 0.0
/msf_viconpos_sensor/pose_sensor/init/p_ic/x: 0  
/msf_viconpos_sensor/pose_sensor/init/p_ic/y: 0 
/msf_viconpos_sensor/pose_sensor/init/p_ic/z: 0 

/msf_viconpos_sensor/pose_sensor/pose_fixed_scale: true
/msf_viconpos_sensor/pose_sensor/pose_fixed_p_ic: true
/msf_viconpos_sensor/pose_sensor/pose_fixed_q_ic: true
/msf_viconpos_sensor/pose_sensor/pose_fixed_p_wv: true
/msf_viconpos_sensor/pose_sensor/pose_fixed_q_wv: true
 
/msf_viconpos_sensor/pose_sensor/pose_absolute_measurements: true
/msf_viconpos_sensor/pose_sensor/pose_use_fixed_covariance: true
/msf_viconpos_sensor/pose_sensor/pose_measurement_world_sensor: false  # selects if sensor measures its position w.r.t. world (true, e.g. Vicon) or the position of the world coordinate system w.r.t. the sensor (false, e.g. ethzasl_ptam

/msf_viconpos_sensor/pose_sensor/pose_measurement_minimum_dt: 0.0  # Sets the minimum time in seconds between two pose measurements.
```

We would really appreciate your input. Thanks in advance!

simonlynen, 2015-11-10:
You will need to set: /msf_viconpos_sensor/pose_sensor/pose_measurement_world_sensor = true

Also I assume that you directly provide the pose of the IMU to the system as position input?

boeboe26, 2015-11-11:
Thanks for your rapid answer!

Now, I set /msf_viconpos_sensor/pose_sensor/pose_measurement_world_sensor = true and the result looks better indeed:

[image]

However, some zigzagging still remains. Considering that the input is ideal, how could this be explained?
[image]

I provide Imu() and TransformStamped() messages as input: imu linear accelerations and angular velocities, vision rotation and translation, just like in the Viconpos example...?

simonlynen, 2015-11-11:
How are the position measurements generated? Do both the position signal
and the acceleration signal come from the simulation framework you use?
Does the integrated accel correspond to the position measurement you
provide? How do you initialize the orientation?

simonlynen, 2015-11-11:
Also what is the rate of your position measurements?

boeboe26, 2015-11-11:
Thank you for your questions, I found the bug. The acceleration sensor coordinate system was turned in order to change the sign of gravity (unfortunately not visible in the plot that I have shown here...). The framework converges to the ground truth very quickly:

[image]

Thanks again!

closed 2015-11-11 by boeboe26