ethz-asl/ethzasl_msf #130: scale estimation for monocular slam

Changliu52, 2015-12-02:
Dear author,

Since we don't have to define the system noise for scale factor when running the code to estimate the visual scale, I guess it was hard coded somewhere in the source code. Could you please point me what is this assumed value?

Kind Regards,

Chang

simonlynen, 2015-12-02:
See here: https://github.com/ethz-asl/ethzasl_msf/blob/master/msf_updates/pose_sensor_fix.yaml#L60

Changliu52, 2015-12-02:
Hi Simon,

Thank you very much for fast reply.
I am expecting a non zero value in order to perform scale estimation. Could you give some suggestion on how to choose that value?

Kind regards
Chang

simonlynen, 2015-12-02:
You don't need to artificically inflate the covariance of the scale parameter to allow estimation of it. You should be fine running the system with a value of zero. You can consider increasing the value or setting an larger uncertainty in the initial covariance if you are far from the true scale at initialization.

Changliu52, 2015-12-02:
Cool. Thank you very much Simon. That's very useful!

Chang

closed 2015-12-03 by Changliu52