ethz-asl/ethzasl_msf #132: Sub/Super script mismatch in pose_measurement.h?

versatran01, 2015-12-04:
In 
https://github.com/ethz-asl/ethzasl_msf/blob/master/msf_updates/include/msf_updates/pose_sensor_handler/pose_measurement.h

line 313 when you calculate q_err:

q_err = (state.Get<StateQwvIdx>()
          * state.Get<StateDefinition_T::q>()
          * state.Get<StateQicIdx>()).conjugate() * z_q_;

This looks like you are doing `(q_wv * q_wi * q_ic)^-1 * z_q_vc`, but should it be `(q_vw * q_wi * q_ic)^-1 * z_q_vc`?

v0n0, 2016-01-19:
I looked multiple ties at your two formulas, but they look exactly the same?

versatran01, 2016-01-19:
@v0n0 the first quaternion is questionable.
In their code it is `q_wv` that represents a rotation from vision to world, the one I put is `q_vw`, which represents the opposite rotation.

v0n0, 2016-01-20:
Oh yeah, just saw that. These indexes are quite confusing, `c = camera` is used, then `v = visual` is used for another, aren't they the same thing? If the variable names were written fully like `quaternion_world_to_vision`, it would be much simpler and more readable and there would be less errors like this.

versatran01, 2016-01-20:
Yes I agree that they are confusing, but more importantly, this code might be wrong.
`q_wv * q_wi` is meaningless, while `q_vw * q_wi` gives you `q_vi`.
Just curious how they get it to work with a wrong expression.

simonlynen, 2016-01-20:
Please look at https://github.com/ethz-asl/ethzasl_msf/pull/116

versatran01, 2016-01-20:
@simonlynen Thanks.

So this is fixed in that branch but not the master one?

closed 2020-12-09 by versatran01