ethz-asl/ethzasl_msf #144: Recommended background reading to error state KF theory

NikolausDemmel, 2016-05-11:
I was wondering if someone could suggest reading material for the theoretical background on error state Kalman filters. I can follow most explanations (I think) of the ideas implemented in MSF (some at least at an intuitive level), but what I have trouble with in particular in the works describing MSF are the derivations of the discrete and linearized error state transition and measurement equations.

What I am looking for is 
* a treatment on error state KF and its relation to EKF and linearized KF
* a formalism / methodology how I get from non-linear, full-state, continuous time system dynamics and measurment equations to discrete time, error state linearizations that I can the implement e.g. in a custom sensor "plugin" or adapted state transition.

What is referenced for these derivations in @stephanweiss 's thesis seems to be http://www.amazon.de/dp/0124807011/ref=wl_it_dp_o_pC_nS_ttl?_encoding=UTF8&colid=1LIBKE0BY45TK&coliid=I363ALQSP5N6BH. Would you recommend that as an entry point? What do others think @simonlynen, @markusachtelik, @burrimi, @ffurrer ?

simonlynen, 2016-05-12:
Hi Niko, 

Maybeck is definitely a very good reference. In terms of getting the basic understanding of an indirect EKF, you might want to read: "[Circumventing dynamic modeling: Evaluation of the error-state kalman filter applied to mobile robot localization](https://trac.v2.nl/export/6833/andres/Documentation/INS%20Kalman/Dynamic%20model%20error%20state.pdf)". 

Nikolas Trawny's TR is a good reference for the [Indirect Kalman filter for 3D attitude estimation](http://www-users.cs.umn.edu/~trawny/Publications/Quaternions_3D.pdf). 

Once you are done with these you can take a look at Joel's IJRR [Camera-IMU-based Localization: Observability Analysis and Consistency Improvement](http://www-users.cs.umn.edu/~dkottas/pdfs/ijrr-oc-vins.pdf)

NikolausDemmel, 2016-05-17:
Hi Simon,

thanks for the recommendations!

I ordered myself a copy of the Maybeck; that should make for some good bed-time reading for the coming weeks ;-).

I have read "Circumventing dynamic modeling: Evaluation of the error-state kalman filter applied to mobile robot localization" before and it is a good paper to get the general idea behind using indirect Kalman filtering for localization, but doesn't really help with understanding the theory thoroughly.

I was aware of "Indirect Kalman Filter for 3D Attitude Estimation" and it is on my list for in-depth review, but I just very recently came across Joan Sola's ["Quaternion kinematics for the error-state KF"](http://www.iri.upc.edu/people/jsola/JoanSola/objectes/notes/kinematics.pdf), which claims to fulfil a similar purpose to Trawny's tech-report, but it seems to be a more modern version, taking care in particular to carefully discuss the different quaternion conventions found in current literature and software libraries. So I think I will next work through the Sola, and then check if the Trawny contains anything on top top of that.

As for Joel's journal paper, I hadn't yet looked at it. From glancing it over, it seems that it does not cover the basics, but provides a treatment of how to analyse observability and how to ensure that the linearised filter has the same observability properties and thus does not become inconsistent as easily. But AFAICT one could do completely without that extension to start with, albeit presumably with reduced consistency and accuracy. Does that make sense?

One more thing, since Joel mentions in that paper that this work is more general than the "MSCKF 2.0" (closed-form error transition matrix computation, orientation error re-parameterization in global coordinates, first-estimate Jacobians, and camera-to-IMU calibration as part of the state) enhancements by Li and Mourikis ("Improving the accuracy of EKF-based visual-inertial odometry" and later "High-precision, consistent EKF-based visual-inertial odometry"), would you agree that Joel's method makes the enhancements described by Li obsolete, or should they be combined?

Lastly, so far I have only come across indirect KF for localization in a context where IMU is used for prediction. Do you know if the error state formulation is also common in other cases?

benjaminabruzzo, 2018-07-17:
I know this is an old thread, but I found Jay Farrell's [Aided Navigation](https://www.amazon.com/Aided-Navigation-High-Rate-Sensors/dp/0071493298) to be very useful.  Though, it is a more application based text.