ethz-asl/ethzasl_msf #147: Getting it working with ORB_SLAM

bishwa9, 2016-06-29:
Hello,

I am trying to get it to work with ORB_SLAM. I am providing the vslam updates (p and q of camera w.r.t. the visual world) at ~20Hz. I am providing imu updates (just rotation vel and lineat acc.) at 40Hz. 

My problems:
Scale drops quickly
Scale drops to negative
There is a pose throttle error.
[image]

If I move the IMU+camera setup, the values don't drop as quickly.

This is my config yaml file:
[image]

This is my launch file:
[image]

As you can see in the yaml file, I am providing the p_ic and q_ic. The q_ic I get from the Rotation matrix from the imu frame to the camera frame. I keep these fixed. Also, my vslam world isn't aligned with the inertial world (NED). So, I exposed q_wv and p_wv. I get q_wv from the Rotation matrix from intertial world to camera world. Since, I start the msf when the camera is at origin of slam world, I put 0,0,0 for p_wv. 

Please help.

YonatanSimson, 2016-07-27:
How do you get outputs from ORB SLAM? I had trouble getting the ROS example in ORB SLAM2 to work. Are you working with ORB SLAM Mono version 1?

bishwa9, 2016-07-27:
No, I am using ORB_SLAM 2. I wrote my own publisher that publishes a geometry_msgs::TransformStamped message. It works but my problem is #148 .

lligen, 2016-08-23:
wonder if you have fix you problem , i am trying to  run ORB_SLAM 2 with this package  but have faced some problem.

mhzaki, 2016-10-10:
Hi @bishwa9 

I want to fuse the ORB SLAM with the IMU data using the datasets available in http://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets but unfortunately i am beginner with ROS and i can't write the publishers needed by the ethzasl_msf package. 

please if you can help me .

Thank you .

bishwa9, 2016-10-10:
@mhzaki So, can you give me a systems overview? Basically, do you have ORB_SLAM running and providing pose outputs? Or do you want to use the Stereo images in the dataset to run ORB SLAM on and then fuse it with the IMU data available in the dataset?

Basically, if you have a separate ORB_SLAM, they have a ROS node written, which should be publishing 6DOF pose. The dataset has IMU readings. Look at msf_updates/launch/pose_sensor.launch. This launch file exposes the two topics that you need to remap. Instead of "/bluebird/fcu/imu" write your imu topic and instead of "vslam/pose" write the ORB_SLAM's output topic. Make sure the topic types are the same. Once done you should be able to just launch your IMU node, ORB node, and then this launch file and get results. Now, I never got it to stabilize with ORB, Refer to #148 because there someone mentioned that they have ORB working with this framework. 

Hope this helps!

mhzaki, 2016-10-10:
hi @bishwa9 
thank you for these informations i didn't know them,
so, my system overview is as you say i want to 
1- run the ORB SLAM withe the a monocular images;
2- fuse it with the IMU data available in the dataset;
3- to have at the same time the scale estimation and the EKF corrected pose.

my problem is i don't know at time how to write publishers witch publish for me the ORB pose, read the IMU data and publish them in a topic .

thanks for you again.

bishwa9, 2016-10-11:
@mhzaki so the dataset has rosbag format, meaning if you run the rosbag file you will be getting the imu data in a topic you will just have to write the topic name in the launch file mentioned in the previous comment. As for publishing and subscribing take a look at ROS tutorials.

mhzaki, 2016-10-13:
Thank you so much, i hope that works with your benefits helps.

mhzaki, 2016-10-14:
Hi @bishwa9  thank you so much your help was great, i can now publish the ORB pose in ORB_SLAM/pose topic and the rqt_graph seem normal but my problem now is i have no data a the plot revelant command also i have an error in the ORB SLAM window (take a look in images plz)
[image]
[image]

thank you again.

LiangLiang128, 2018-05-06:
Hello,@mhzaki,Did you have solved these problems you mentioned above?
1- run the ORB SLAM withe the a monocular images;
2- fuse it with the IMU data available in the dataset;
3- to have at the same time the scale estimation and the EKF corrected pose.
I have the same problems with you.If you have succeed,please help me.:)Thanks a lot.

nalinraut, 2019-03-26:
@mhzaki If you have successfully implemented the code, is it possible for you to share the same? I am trying to something similar to implement SLAM on mobile phone using ORB and IMU.