ethz-asl/ethzasl_msf #163: Message synchronization

baryn, 2017-05-22:
Hi,

First of all, thanks a lot for sharing this amazing work. I have been able to run the tutorial with the dataset.bag and now I am trying to use my own containing two topics populated with pose measurements (30Hz) and imu readings (150Hz) respectively but I get this error msg after a few seconds:

**[ERROR] [1495464561.512402042]: ProcessIMUlatest IMU message was out of order by a too large amount, resetting EKF: last-state-time: 1548.29 current-imu-time: 1544**
and the process stops, no new msgs are published on the topic /msf_core/pose for example.

Does it mean my IMU msg is to much delayed ? Is there something about clock synchronization somehow ? Note that both my topic headers start with the same ros time stamp and I have no data gaps.
I would greatly appreciate some help on this one.

PS: extra question: It is written on the tutorial that the _/msf_core/pose_ msg which is published at IMU rate should be used for robot control, but I would like to have the full estimated state (p,p_dot,R,R_dot) published at IMU rate. The _/msf_core/state_out_ msg does this job but is it published at pose rate (30Hz here), too low for robot control. Am I missing something here ?

Thanks a lot.

IamARR, 2018-10-02:
Hi,

Did you achieved the solution of this error? 
If the answer is yes, I would like to know how to fix it.

Thanks a lot.

baryn, 2018-10-03:
Hi,

No, I didn't find a way to solve this. I haven't worked on it for a while now. Sorry, I cannot help you.

Cheers.

MohammadKassemZein, 2020-07-14:
Did Anybody solve this issue? Any help would be really appreciated!

closed 2023-12-15 by baryn