ethz-asl/ethzasl_msf #165: message drop curr and large time-gap

fx815, 2017-07-14:
Hello,
I'm using vicon pose sensor and exactly the same parameters in viconpos_sensor_fix.yaml.
When I initialize the scale. I got some warnings as following:

[ INFO] [1500050513.495970190]: Initialize filter with scale 1.000000
[ INFO] [1500050513.496132456]: initial measurement pos:[ 0.452168  0.576429 0.0563536] orientation: [1, 0.00807, 0.00989, 0.00573]
[ WARN] [1500050513.754366679]: Using simulated core plus fixed diag initial error state covariance.
[ INFO] [1500050513.764550252]: Initializing msf_core (built: Jul 14 2017)
[ INFO] [1500050513.764687893]: Core parameters: 
	fixed_bias:	0
	fuzzythres:	0.1
	noise_acc:	0.083
	noise_accbias:	0.0083
	noise_gyr:	0.0013
	noise_gyrbias:	0.00013

[ INFO] [1500050513.764981585]: Core init with state: 
--------- State at time 513.654s: ---------
0 : [0-2]	 : Matrix<3, 1>         : [ 0.452168  0.576429 0.0563536]
1 : [3-5]	 : Matrix<3, 1>         : [0 0 0]
2 : [6-9]	 : Quaternion (w,x,y,z) : [1, 0.00807, 0.00989, 0.00573]
3 : [10-12]	 : Matrix<3, 1>         : [0 0 0]
4 : [13-15]	 : Matrix<3, 1>         : [0 0 0]
5 : [16-16]	 : Matrix<1, 1>         : [1]
6 : [17-20]	 : Quaternion (w,x,y,z) : [1, 0, 0, 0]
7 : [21-23]	 : Matrix<3, 1>         : [0 0 0]
8 : [24-27]	 : Quaternion (w,x,y,z) : [1, 0, 0, 0]
9 : [28-30]	 : Matrix<3, 1>         : [0 0 0]
-------------------------------------------------------
[ WARN] [1500050513.765169601]: large time-gap re-initializing to last state

[ WARN] [1500050513.765314835]: /flourish/vrpn_client/raw_transform: message drop curr seq:35827 expected: 35820
[ INFO] [1500050513.765562143]: Control interface got first odometry message.
[ WARN] [1500050535.094545511]: Pose measurement throttling is on, dropping messagesto be below 20.000000 Hz
[ WARN] [1500050565.111835547]: Pose measurement throttling is on, dropping messagesto be below 20.000000 Hz
[ WARN] [1500050595.122652877]: Pose measurement throttling is on, dropping messagesto be below 20.000000 Hz
[ WARN] [1500050625.123100924]: Pose measurement throttling is on, dropping messagesto be below 20.000000 Hz
[ WARN] [1500050655.123158141]: Pose measurement throttling is on, dropping messagesto be below 20.000000 Hz
STATUS armCallback,line 199: Need to obtain control first, 0x0001
[ INFO] [1500050674.988717526]: keyboard callback
Success in obtaining control!!
STATUS setControlCallback,line 902: Obtain control failed: switch to F mode
[ INFO] [1500050683.727305120]: keyboard callback
[ INFO] [1500050683.984370165]: keyboard callback
[ WARN] [1500050685.137651924]: Pose measurement throttling is on, dropping messagesto be below 20.000000 Hz
[ WARN] [1500050697.809234190]: /flourish/vrpn_client/raw_transform: message drop curr seq:54033 expected: 54032
[ WARN] [1500050708.637581655]: /flourish/vrpn_client/raw_transform: message drop curr seq:55081 expected: 55030
[ WARN] [1500050715.142157877]: Pose measurement throttling is on, dropping messagesto be below 20.000000 Hz
[ WARN] [1500050719.297265649]: /flourish/vrpn_client/raw_transform: message drop curr seq:56148 expected: 56147
[ WARN] [1500050719.492901516]: /flourish/vrpn_client/raw_transform: message drop curr seq:56164 expected: 56163
[ WARN] [1500050728.302946423]: /flourish/vrpn_client/raw_transform: message drop curr seq:56872 expected: 56871
[ WARN] [1500050728.303012375]: /flourish/vrpn_client/raw_transform: message drop curr seq:56901 expected: 56873
[ERROR] [1500050728.303085270]: GetClosestState Requested closest state to 727.15 but there was no suitable state in the map
[ERROR] [1500050728.303142465]: AddMeasurement GetClosestState returned an invalid state
[ERROR] [1500050728.303178753]: No measurement was applied, this should not happen.
[ WARN] [1500050728.303219966]: /flourish/vrpn_client/raw_transform: message drop curr seq:56912 expected: 56902
[ WARN] [1500050728.315921604]: /flourish/vrpn_client/raw_transform: message drop curr seq:56958 expected: 56934
[ WARN] [1500050728.652883996]: /flourish/vrpn_client/raw_transform: message drop curr seq:57009 expected: 56980
[ WARN] [1500050745.157202967]: Pose measurement throttling is on, dropping messagesto be below 20.000000 Hz
[ WARN] [1500050753.947049201]: /flourish/vrpn_client/raw_transform: message drop curr seq:59512 expected: 59511
[ WARN] [1500050775.166802989]: Pose measurement throttling is on, dropping messagesto be below 20.000000 Hz
[ WARN] [1500050805.169738986]: Pose measurement throttling is on, dropping messagesto be below 20.000000 Hz

I sync the time between my ground station and onboard computer everytime when I open the computers. The question is that what's going wrong and would these affect the odometry I get and how to correct these.

Thank you very much!

Feng

MohammadKassemZein, 2020-07-20:
@fx815 I am facing the same issue. How did you manage to sync the tume between your ground station and the onboard computer?

fx815, 2020-07-30:
> @fx815 I am facing the same issue. How did you manage to sync the tume between your ground station and the onboard computer?

Hi, I used chrony as a time sync tool.

MohammadKassemZein, 2020-07-30:
@fx815 Thank you for response! 
Could you please tell me what are the steps?
Thank you in advance!

fx815, 2020-08-12:
> @fx815 Thank you for response!
> Could you please tell me what are the steps?
> Thank you in advance!

Hi, it's essentially set one computer in your ros network as time server and the others as client. Installing and configuring chrony can do that, please google that, there are lots of resources.