ethz-asl/ethzasl_msf #203: 10-year anniversary of SSF/MSF

stephanweiss, 2022-12-12:
@ current SSF and MSF maintainers and developers:

I recently see increased activity on MSF again and it filled me with great joy that a 10 year old framework we developed back then with @simonlynen, @markusachtelik, and @laurentkneip is still in use for research! Thanks much!

That being said, 10 years in research is quite a time and since more than a year there is the newer modular MaRS (https://github.com/aau-cns/mars_lib) sensor fusion framework out that alleviate several issues of SSF & MSF (including online sensor integration, initialization methods, multi-sensor scalability, etc). Like MSF, MaRS builds on SSF and should allow for simple usage and extensions.

We can still keep/maintain SSF/MSF but maybe we should add a link to MaRS on the ROS wiki and Github pages so the community can use the next generation of multi-sensor fusion? As one of the authors of SSF/MSF I think it is about time for such a switch.

Looking forward to your feedback,
Stephan

rikba, 2022-12-15:
Hi @stephanweiss 

This is great. I added a link on the front page. https://github.com/ethz-asl/ethzasl_msf/blob/master/README.md#other-frameworks

Is this what you had in mind?

closed 2022-12-15 by rikba

stephanweiss, 2022-12-19:
@rikba, fantastic, thanks much!
Should we add something on the ROS pages as well?
SSF: http://wiki.ros.org/ethzasl_sensor_fusion
MSF:  http://wiki.ros.org/ethzasl_sensor_fusion/Tutorials/Introductory%20Tutorial%20for%20Multi-Sensor%20Fusion%20Framework